DavidArmstrong/SCL3300 #5: Connection issues - crcerr

janz0, 2020-03-02:
I've tried with a couple different boards, a couple esp32 boards (huzzah32), uno (with logic level converter). I seem to receive the crc error with all transmissions as a result I am unable to connect to the SCL3300 properly.

Do you have any tips to obtain a connection? 

Below is an attempt to connect (including debug data):
15:00:06.986 -> Reading Raw register values from SCL3300 Inclinometer
15:00:06.986 -> B4000338 B4 0 3 38  3 0 82 C  1 82 C 61 1
15:00:07.035 -> 180000E5 18 0 0 E5  2 8 20 83  1 820 83 2F 1
15:00:07.035 -> 180000E5 18 0 0 E5  2 8 20 83  1 820 83 2F 1
15:00:07.035 -> 180000E5 18 0 0 E5  2 8 20 83  1 820 83 2F 1
15:00:07.035 -> B0001F6F B0 0 1F 6F  2 8 20 C0  1 820 C0 2F 1
15:00:07.035 -> Murata SCL3300 inclinometer not connected.

A great library once I have the SPI transmitting and receiving properly I think.

janz0, 2020-03-02:
I should also note that I'm using the evaluation pcb. I've tested the two pcb's I have with the muRata evaluation unit and had no issues.

DavidArmstrong, 2020-03-02:
Do you have a pullup resistor on the SPI slave select line? I found some
boards require it to get SPI to work at all. Others are fine without it.

Dave

janz0, 2020-03-03:
That was going to be my attempts tomorrow.

In the specs it says something along the lines of “If you receive a checksum error on MISO then there are communication issues.” I’m hoping that is my issue and I’ll have it working after this!

If you don’t mind my asking, what boards/setup are you using for testing?

Mike

DavidArmstrong, 2020-03-03:
I've used an Arduino Pro mini and a Sparkfun Redboard Turbo for testing. Be
careful of using 1/8th watt resistors on a breadboard, as the thin leads
don't make a good connection.

Another useful test is a loop back test. Take out the inclinometer and tie
MISO and MOSI together on the SPI interface. The library will still 'fail'
because the whoami register won't be the expected value. But CRC errors
should go away.

Dave

DavidArmstrong, 2020-03-03:
Here is a comparison of 'good' debug data being output:
Reading Raw register values from SCL3300 Inclinometer
B4000338 B4 0 3 38  FF 0 0 FF  1 0 FF FF 0
180000E5 18 0 0 E5  B7 0 3 B4  1 3 B4 B4 0
180000E5 18 0 0 E5  1B 0 40 7A  1 40 7A 7A 0
180000E5 18 0 0 E5  19 0 40 79  1 40 79 79 0
B0001F6F B0 0 1F 6F  19 0 0 6A  1 0 6A 6A 0
40000091 40 0 0 91  B1 0 1F E0  1 1F E0 E0 0
40000091 40 0 0 91  41 0 C1 36  1 C1 36 36 0
FC000073 FC 0 0 73  41 0 C1 36  1 C1 36 36 0
40000091 40 0 0 91  FD 0 0 FC  1 0 FC FC 0
40000091 40 0 0 91  41 0 C1 36  1 C1 36 36 0
Inclinometer is still alive...

The last number in each line tells whether or not the CRC value is valid, zero is good, a '1' isn't.  The two sets of hex numbers before that tell you the reported CRC that was in the 32-bit word returned, and the computed CRC.  They should always be the same value.  If not, then something is wrong in the SPI communication channel.

janz0, 2020-03-03:
You are correct, it was an issue with the spi communications. I did the following as tests:
1. Connected the MOSI directly to the MISO on my esp32 to check the board was communicating, no errors.
2. Added code to allow for three attempts to startup the inclinometer "inclinometer.begin()"
3. Added a pull up resistor on the chip select. This decreased the checksum errors, intermittently the examples would pass into readings.
4. Added a pull down resistor on the MISO line. This further decreased errors, example scripts typically operated. After a random interval it would still fail the WHOAMI check while grabbing data and would not recover.
5. Complete redesign with a larger bread board and better connectors. Example scripts operated normally without 1 through 4 required.

I believe I must have had a cheap/faulty connector in the mix that was disrupting my connection. All is operating as expected now.

Your library came a the perfect time for myself, I'll let you know if I come across any other issues.

closed 2020-03-03 by janz0