DavidArmstrong/SCL3300 #8: Reading accelerometer values

LouisR-git, 2020-05-19:
Hello,

I use the Murata PCB SCL3300 and an Arduino MKR WIFI 1010.

It seems to me that there is a problem with the values of the accelerometer.
Serial monitor with "Example3_BasicAccelerometerReading" program :
`X Accelerometer: 0.01	Y Accelerometer: 5.44	Z Accelerometer: 0.99`

The value of the y axis is false.

I think the problem is the variable type for acceleration datas.
The datasheet specifies that values are in complement format so you should declare signed int instead of unsigned int.

The modifications are to be made in : 
struct of SCL3300data
the fonction `double acceleration(int16_t ACC)`

Thank you so much for this library which is of the greatest help !

DavidArmstrong, 2020-05-20:
You are quite correct.  I was also able to find some related changes that needed to be made.  Will update the library shortly.

DavidArmstrong, 2020-05-20:
code changes committed and merged

closed 2020-05-20 by DavidArmstrong