DavidArmstrong/SCL3300 #10: Operator error

Douglasov, 2020-07-27:
Hello, this function

//Return value of Error Flag 1 register
uint16_t SCL3300::getErrFlag1(void) {
  beginTransmission(); //Set up this SPI port/bus
  transfer(SwtchBnk0);
  transfer(RdErrFlg1);
  transfer(RdErrFlg1);
  endTransmission(); //Let go of SPI port/bus
  if (crcerr || statuserr) return (uint16_t)(CMD && 0xff); //check CRC and RS bits
  return DATA;
}

the operator && maybe is &?

DavidArmstrong, 2020-07-28:
I will review this.

Douglasov, 2020-07-28:
There is another parts of the code with same change of operator, and one part where is dislocated 16 bytes in a variable of 16 bytes.

DavidArmstrong, 2020-07-29:
I am examining each instance and will need to do regression testing before releasing any updates.  This will most likely be implemented as separate updates, rather than rushing to get them all handled at once.

closed 2020-07-29 by DavidArmstrong

Douglasov, 2020-07-29:
In this part of code 
serialNum = (DATA << 16) | serialNum;
A uint_16 has shift by 16 so have no effect. Another thing them I sse after changes is the library on works again after a hard change of mesuares after use a function begin().

DavidArmstrong, 2020-07-29:
serialNum is a 32-bit number, so DATA is being converted from 16 bits to 32 bits.  It takes two 16-bit data reads to get the 32-bit result for the Serial Number of the part.  So this is correct, and it outputs the correct value as seen in the Example5_RawDataReading.ino sketch.

Douglasov, 2020-07-29:
I don't see this sorry