DavidArmstrong/SCL3300 #12: Error checking redesign

DavidArmstrong, 2020-07-29:
The current means of checking for errors when receiving a response from the SCL3300 is really inadequate, for lack of a better word.  While running some tests to improve this functionality, I discovered that the current design won't allow for modifications on the error checking code without adverse effects being seen.  So, to avoid this, I need to re=code the error checking portions, while maintaining all the working functionality and compatibility.
Therefore, I'm opening this issue to track this change.  As stated, there will be no change in functionality, or in calling the library as it stands.  However, if one needs to check to see if there was an error reported by the SCL3300, that will be explicitly laid out for the user to include in their usage of the library.

JLH-94, 2020-08-04:
Hello,
i am trying to use your SCL3300 library with the "new" PortentaH7 from Arduino. Unfortunately I get a longer error message when compiling. It seems to be about the CRC calculation. Maybe you could have a look at it, before i try to reprogram a complete new library.

I postet a request in the PortentaH7-Forum:
[https://forum.arduino.cc/index.php?topic=698825.0](url)

Thanks for your effort with that lib. in any case.

DavidArmstrong, 2020-08-11:
After going through the library and updating how error checking is done and handled, I came to the realization that to 'do the right thing', I have to introduce an incompatibility, however small.  I really wanted to avoid this, but handling error conditions is an important enough reason to make a change of this nature.

This change will force the revision number to go to 3.0.0.

The actual code changes are relatively minor, and the change needed to previous sketches is minor too.  It can be limited to one changed line in the sketch.

Here is an example...
Version 2.1.4 would have:
if (inclinometer.available()) {
  //Do stuff here
}

And for Version 3.0.0, this becomes:
if (inclinometer.available()) {
  //Do stuff here
} else inclinometer.reset();

All that is needed is the extra 'else' clause at the end of the 'if' structure when used with the 'available()' function.  This will do a software reset of the SCL3300 sensor should an error condition be detected.  However, if old sketches are not updated, and the reset isn't done when the SCL3300 enters an error state, then the sensor will stay in that error state, and available() will never return a 'true' value again.  (At least, until a power cycle is done, or until reset()/begin() calls are made.)

closed 2020-08-11 by DavidArmstrong