DavidArmstrong/SCL3300 #16: SPI VSPI setting CS Issue

Andyhast39, 2020-09-20:
I'm probably barking up the wrong tree here but I've had an SCL3300 working very well off an ESP32 Wroom. I've moved over to the next iteration of my project and I had to move the MISO/MOSI/SCK/CS pins and now it refuses to respond. 
Its MISO is on the VSPI GPIO 19, CLK GPIO 18, CS GPIO 32. I have changed the SCL3300.h to change the CS pin to 32, but I'm guessing its the default SPI settings? 
Can you help? I think I'm just missing one vital detail.....

Cheers,

Andy

DavidArmstrong, 2020-09-22:
If you are only moving the chip select or slave select pin, a different digital output pin should be able to be selected by using the "begin(32)" function call at the start of your program, for Arduino pin 32.  The current design uses whatever the default pins are for SPI for MISO, MOSI, and SCK.  The select pin is easily switched to any other available digital output.

Of course, the pin numbering on the Arduino IDE does not always follow what one would expect.  For example, on a SAMD21 M0 processor, the MPU pin 27 corresponds to D10 on the Arduino header.  So one has to check what the Arduino IDE core definition files have for the pin in question.

However, if you are attempting to use a completely different set of pins for the SPI interface, not just chip select, that is a different issue.

Andyhast39, 2020-09-24:
Thanks David. After a lot of trial and error I was indeed missing a very vital detail. The chip turned out to be an SCA3300 and not the expected SCL3300 and as such would never respond as expected.
Thanks for you help.

closed 2020-09-24 by Andyhast39