DavidArmstrong/SCL3300 #21: SPI Issues

NeelMani3, 2020-10-19:
First of all, thanks for the library. It is very handy and useful. I am trying to connect SCL3300-D01-PCB to arduino uno through a bidirectional logic converter. Also using a pull up resistor for chip select - 4.7K. DVIO pin is at +3.3 Volts. I have checked all the connections but still I am not getting the communication established. I have provided the serial outputs with debug. 

[image]

I have also added more serial prints in the code (below output) and still not able to figure out the issue.

14:33:39.307 -> Reading basic Tilt Level Offset values from SCL3300 Inclinometer
14:33:39.384 -> FC000073  Inside transfer I FC 0 0 73 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:39.502 -> B4002098  Inside transfer I B4 0 20 98 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:39.618 -> B4000338  Inside transfer I B4 0 3 38 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:39.739 -> 180000E5  Inside transfer I 18 0 0 E5 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:39.868 -> 180000E5  Inside transfer I 18 0 0 E5 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:39.989 -> 180000E5  Inside transfer I 18 0 0 E5 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:40.101 -> B0001F6F  Inside transfer I B0 0 1F 6F In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:40.244 -> 40000091  Inside transfer I 40 0 0 91 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:40.358 -> 40000091  Inside transfer I 40 0 0 91 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:40.484 -> B4000338  Inside transfer I B4 0 3 38 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:40.605 -> 180000E5  Inside transfer I 18 0 0 E5 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:40.725 -> 180000E5  Inside transfer I 18 0 0 E5 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:40.841 -> 180000E5  Inside transfer I 18 0 0 E5 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:40.968 -> B0001F6F  Inside transfer I B0 0 1F 6F In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:41.085 -> 40000091  Inside transfer I 40 0 0 91 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:41.202 -> 40000091  Inside transfer I 40 0 0 91 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:41.320 -> getErrFlag1
14:33:41.320 -> FC000073  Inside transfer I FC 0 0 73 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:41.447 -> 1C0000E3  Inside transfer I 1C 0 0 E3 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:41.600 -> 1C0000E3  Inside transfer I 1C 0 0 E3 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:41.712 -> 65535
14:33:41.712 -> getErrFlag2
14:33:41.712 -> FC000073  Inside transfer I FC 0 0 73 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:41.838 -> 200000C1  Inside transfer I 20 0 0 C1 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:41.960 -> 200000C1  Inside transfer I 20 0 0 C1 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
14:33:42.092 -> 65535
14:33:42.092 -> Murata SCL3300 inclinometer not connected.

NeelMani3, 2020-10-19:
Just wanted to add that I also did try the loopback test (that has been suggested for the other connectivity issue). 

When I loop back the CRC errors go off 
18:12:00.841 -> Reading basic Tilt Level Offset values from SCL3300 Inclinometer
18:12:00.925 -> FC000073  Inside transfer I FC 0 0 73 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
18:12:01.040 -> B4002098  Inside transfer I B4 0 20 98 In CRC TrueIn RS True Inside transfer II Status err1 CRC err1 3 FFFF FF 7A 1
18:12:01.174 -> B4000338  Inside transfer I B4 0 3 38 In CRC FalseIn RS True Inside transfer II Status err1 CRC err0 0 3 38 38 0
18:12:01.289 -> 180000E5  Inside transfer I 18 0 0 E5 In CRC FalseIn RS True Inside transfer II Status err1 CRC err0 0 0 E5 E5 0
18:12:01.396 -> 180000E5  Inside transfer I 18 0 0 E5 In CRC FalseIn RS True Inside transfer II Status err1 CRC err0 0 0 E5 E5 0
18:12:01.506 -> 180000E5  Inside transfer I 18 0 0 E5 In CRC FalseIn RS True Inside transfer II Status err1 CRC err0 0 0 E5 E5 0
18:12:01.622 -> B0001F6F  Inside transfer I B0 0 1F 6F In CRC FalseIn RS True Inside transfer II Status err1 CRC err0 0 1F 6F 6F 0
18:12:01.740 -> 40000091  Inside transfer I 40 0 0 91 In CRC FalseIn RS True Inside transfer II Status err1 CRC err0 0 0 91 91 0
18:12:01.856 -> 40000091  Inside transfer I 40 0 0 91 In CRC FalseIn RS True Inside transfer II Status err1 CRC err0 0 0 91 91 0

but when I am connecting to the inclinometer normally without the loopback I am getting the FF errors...

[image]

DavidArmstrong, 2020-10-19:
The outputs you have shared point to some sort of connection issue. I've
run into a similar issue when I'm using a breadboard with the scl3300, and
some of the wires end up having a problem. So I'd recommend double checking
there first. (The wires can look right, but any corrosion will cause this
problem. )

Dave

NeelMani3, 2020-10-19:
Thanks. I tried reconnecting and checked all the wire connectivity with multimeter between uno - bidirectional converter - Inclinometer - all board to board instead of the wires.  Changed couple of wires as well. Still facing the same issue. 
Is there any way that the Inclinometer inherently is having some component issue? Is there any way to check that?

DavidArmstrong, 2020-10-19:
There is always the possibility of a hardware issue with the scl3300
itself. Another check that could be done is to look at the three SPI line
with an oscilloscope, and see what wave forms they show.

Dave

NeelMani3, 2020-10-20:
Thanks for your suggestion. I reconnected the circuit in a new board with nano and removed the pull up resistor on chip select. It worked fine without the errors. Thanks for your help.

DavidArmstrong, 2020-10-23:
Issue reported as resolved, so closing.

closed 2020-10-23 by DavidArmstrong