DavidArmstrong/SCL3300 #22: Duplicates

phhoef, 2021-02-06:
I do have a strange issue and I even don't know if its related to this lib.
But maybe you guys can help me out.

I have a custom Arduino Due compatible board and the SCL3300 is also mounted on this board.
The board is mounted in a chassis and the chassis lays flat on a leveling table (~0°).

When I measure the angle with function `getTiltLevelOffsetAngleX()` a few times, I get the following output:
```
-0.148315429687500

-0.153808593750000

-0.164794921875000
-0.164794921875000

-0.170288085937500
-0.170288085937500
-0.170288085937500
-0.170288085937500
-0.170288085937500
-0.170288085937500

-0.175781250000000
-0.175781250000000
-0.175781250000000
-0.175781250000000

-0.181274414062500
-0.181274414062500
-0.181274414062500
-0.181274414062500
-0.181274414062500
-0.181274414062500
-0.181274414062500
-0.181274414062500
-0.181274414062500
-0.181274414062500
-0.181274414062500
-0.186767578125000
-0.186767578125000

-0.192260742187500

-0.197753906250000
```

The values are sorted in ascending order. As you can see, I get a lot of duplicates. In fact, there are only about 5-6 unique values. From my point of view, that should never ever happen in real life. Lib version 3.0.0 is used.
The measurements are trigger by a command through the Serial interface from my computer. It's about one measurement per second.
When I rotate to chassis, the value changes and I measure angles about 90° (what I have expected).
But again, when the sensor is not moved and multiple measurements are triggered, I see duplicate values.

Does anybody have an idea, why this could happen?
Thanks a lot!

DavidArmstrong, 2021-02-06:
If the sensor is not moving, it would make sense that it would return the same value each time it was read. Also, the available() function has to be called each time a new set of data is to be read from the sensor. That function reads all the sensor axis values as a group, so that they are consistent. Otherwise the angle functions are just reporting what was in the last data set fetched by available(). Finally, the raw data are 16 bit signed numbers that are internally converted to a floating point representation in degrees. So there is a limit to the resolution that is measured by the sensor.

phhoef, 2021-02-06:
Thanks for your answer and your thoughts.
I am reading the sensor values for each iteration of the Arduino main loop. Every iteration `available()` is called.

In a perfect world, the values are the same when the sensor is not moved, but I was expecting much more noise. Do you really think, it is possible, that the values are so consistent?

That said, I see the angle changes when the sensor moves. It's hard to get an absolute correct angle for reference, but holding the chassis at 45° will also produce an measured angle ~45°.

DavidArmstrong, 2021-02-06:
Actually, this particular inclinometer is very low noise, which is one of
the reasons it is preferred over other similar sensors.  Here is a
reference to noise tests done in comparison to other similar modules:
https://www.yoctopuce.com/EN/article/inclinometer-tests

Dave

phhoef, 2021-02-06:
Thanks for the article. I did a comparison of three sensors a while ago (MPU6050, BNO055, SCL3300) and also came to the conclusion, the the SCL3300 is very accurate. That's why I used it in my custom board design.
https://forum.sparkfun.com/viewtopic.php?f=83&t=52967 

It still feels a bit weird, that the sensor produces the exact same outputs when it is not moved - it's too good to be true, I can't believe it.
That said, I didn't see any misbehavior, it just feels suspicious ...

Do you see a similar behavior of your sensor?
Today, I tested it with the evaluation board (https://www.murata.com/en-eu/products/sensor/design-development/scl3300-d01-pcb) connected to an original Arduino Due. And indeed, it also produces measurements, that appears multiple times.

I have a calibration routine, where the mean value of 10 measurements is used to zero the current measurement.
With two different devices, I got the same offset to the 15th decimal place.
That feels weird, that two different sensors brings identical measurements ...

DavidArmstrong, 2021-02-07:
My sensor has always been stable in it's output, except when one of the electrical connections was flakey.  Hence the need for available() to provide an output flag to indicate if the data is valid or not.

phhoef, 2021-02-07:
Again, thanks for your thoughts.
Actually, I am somewhat reassured now. Especially, as the sensor seems to provide valid angles when moving. Attached is the plot of angles, when laying flat on the table and then tilted to 90° and back to 0°.
Still, I think it's exceptional, that the measured values are so stable and occurring multiple times. I wrote an email yesterday evening to Murata and asked if that is possible and expected. I'll keep you posted ...

phhoef, 2021-02-16:
Hallo @DavidArmstrong,

as I mentioned in my last comment, I contacted Murata. They were very friendly and helped me a lot.
To summarize the conversation, there were 2 outcomes

- Hitting exactly 0° is probably normal noise. I also noticed, that the probability to hit 0° is higher, when zeroing the sensor in the first place.
- Duplicates probably occur, when I read the sensor too fast. Not sure, if it's always the case, as I saw different angles before a duplicated occurred, but in some cases it's possible.

They also shared an example code for STM32 mcu. They kindly permitted me to share the code with you in order to improve your Arduino library. I did a quick comparison of both code bases and didn't found much to improve. But you're much more familiar with your code ... They asked us not to make the example code publicity available.
How can I privately contact you to share the example? Of course, I am happy to help to improve your code.

DavidArmstrong, 2021-02-22:
Private message over Sparkfun Forums sent.

closed 2021-02-25 by phhoef