DavidArmstrong/SCL3300 #23: Assistance debugging?

rachelmh, 2021-03-16:
Hi David,

I wasn't sure how to contact you other than through this method. I have tried using this library with both an Arduino Uno and Mega and neither seem to be working. Perhaps it is an issue with my wiring? I followed the wiring suggestions from here: https://www.mouser.com/datasheet/2/281/pcbspec_scx3300-1568230.pdf

and am using a sparkfun bi-directional level shifter as you recommended.  Here is my current wiring  diagram:

(Lo--Hi) is the bi directional level shifter.

On the left half of the chip I have:
AVSS-GND                                   
XReset -                                      
INT -                                            
CLK 0
DVDD- +3.3V
DVSS - GND
DVSS - GND
NC - 

On the right half of the chip I have
AVSS- GND
AVDD - +3.3V
CSB  -  Lo--Hi--Pin D10 Uno
         |
        5kOhm
         |
       +3.3V

MISO- Lo--Hi -- Pin D12 Uno

MOSI- Lo--Hi--Pin D11 Uno

SCK - Lo--Hi-- Pin D13 Uno

NC-

If you have any suggestions I would greatly appreciate it! Please feel free to email me at rachelmh@mit.edu

DavidArmstrong, 2021-03-17:
Replied in private email, and explained how to get debugging output.

DavidArmstrong, 2021-06-13:
Initial issue was resolved offline long ago.  So closing this issue.

closed 2021-06-13 by DavidArmstrong