DavidArmstrong/SCL3300 #24: Compatability

ralarsen2, 2021-03-31:
Hello, using v3.0.2 I am seeing some odd behavior.  The sensor will report ax, ay, and az readings at 10 hertz (seemingly mode 4) however I am unable to change the mode and the sensor will not return TRUE to the inclinometer.begin() command.  Further, the ax, ay, and az readings are all roughly 1/2 of gravity when pointed downward.

The SCA3300 is connected to a Adafruit Feather with a pullup to CSB.  

Can anyone opine on a likely source of such behavior?  
Thank you

DavidArmstrong, 2021-03-31:
If the begin() function is always returning a false value, that suggests a connection problem, as initial device reads are not being seen with valid checksum values.  I would recommend checking all the physical connections first.
Next would be to enable debug in the library, and run an SPI loop back test without the sensor connected.

ralarsen2, 2021-03-31:
Thank you for your quick review.  I will investigate as you suggest, Cheers, Russell

ralarsen2, 2021-04-04:
Hello,
So using the debugging option I am viewing the values sent and returned from the transfer() command within SCL3300.cpp, specifically the begin() portion.

The crcerr and statuserr both return false however the WHOAMI command is returning 81 and not 193 (0xc1).

I did run a loopback test and the sent value is returned.  The wiring appears valid given the ability to communicate, and a have a pull up 4.7kohm on the chip select, and have 3v3 applied to the DVIO pin.

Might I ask your insight with respect to the WHOAMI not returning 193?

I have tried this on an Arduino UNO as well as a Adafruit Feather.  Both resutl in the same WHOAMI value of 81.

Thank you,
Russell

DavidArmstrong, 2021-04-05:
Hi,

A WHOAMI value of 81 decimal = 0x51 hexadecimal. This is the value for an
SCA3300 accelerometer sensor, not an SCL3300 inclinometer. That would
explain why this library is not working for you.

Dave

ralarsen2, 2021-04-05:
Ok, well at least it is a simple answer.  I do appreciate your assistance.  Cheers and thank you.

ralarsen2, 2021-04-06:
With your help I now have the SCA nearly there.  With some changes to the WHOAMI and the LSB settings it reports to scale.  The final confusion is that it will not run faster than 40 hertz on Modes 1, 2, or 3, which for the SCA is a 70 hertz setting.

Have you ever encountered issues with the mode settings?  For example, I found I have to send the mode setting twice with a delay between them.  On the second go the transfer appears to successfully send the command (the mode commands are the same between the SCL and the SCA chip).  The returned values from the transfer differ from the first attempt to the second.  Being an engineer - but not a software engineer - I am at a disadvantage when it comes to why these return values are different . . . and what that might imply.

I do appreciate your help.

Cheers,
Russell

[cid:75d14d03-1a76-4c85-a555-dfe522250370]

DavidArmstrong, 2021-04-06:
Russell,

I have not tested using other modes besides the default value of 4, as that
is the most commonly needed value. If you look at the begin() function
though, you will see that after the mode is set, it does some dummy read
status fetches, to allow time for the hardware reset done by mode setting
to complete. That would explain why you would have to do it twice, with a
delay. As long as that required delay is accounted for in the code, it
should be fine.

Dave

DavidArmstrong, 2021-06-13:
No recent responses on this thread, so closing.

closed 2021-06-13 by DavidArmstrong