DavidArmstrong/SCL3300 #25: Can I use this library for two SCL3300 sensors connected to the same Arduino board?

vpougatchev60, 2021-06-13:
Hi David,

I wonder if there is a way I can use your library in my project where I need to connect two SCL3300 sensors to the same Arduino board. Any suggestions are greatly appreciated. I can be reached at vpougatchev@gmail.com if needed.

Best regards,

Vadim

DavidArmstrong, 2021-06-13:
I would think that this can be done.  The MISO, MOSI, and SCLK signals would be common with both boards.  The Chip Select line would have to be different between the two devices, of course.  So if pins 10 and 9 were used for the two device select lines, then the two instances would be created in the code as follows:

[twodevices.txt](https://github.com/DavidArmstrong/SCL3300/files/6643173/twodevices.txt)

(Note: this is UNTESTED, but based on Example #1 in the library.)
Go ahead and give this a try, and let me know if it works for you.

vpougatchev60, 2021-06-14:
Hi Dave,

 

Thank you very much for your swift reply and suggestions. That is what I thought indeed, but your recommended code will be really helpful to me. I will give it a try and post it in the issue thread that I initially created in your GitHub repo.

 

-----

 

Best regards,

 

Vadim Pougatchev

 

The content of this email is confidential and intended for the recipient specified in the message only. It is strictly forbidden to share any part of this message with any third party, without the written consent of the sender. If you received this message by mistake, please reply to this message and follow with its deletion, so that we can ensure such a mistake does not occur in the future.

vpougatchev60, 2021-08-04:
I finally had a chance to finish the hardware part of our planned design with two SCL3300 sensors and test the recommended code. Everything works very well. Thank you, David, for your help.

closed 2021-08-04 by vpougatchev60