DavidArmstrong/SCL3300 #27: Using the SCL3300 with ESP32

atlaste, 2021-07-26:
I've been looking into using the SCL3300 with an ESP32. The ESP32 provides 3.3V directly, so that shouldn't be a problem. However, I am attempting to figure out what a PCB should look like exactly.

Unfortunately the pre-soldered PCB's are not available on this side of the world for reasonable prices, so I have to fab one. That's not an issue per-se; I found the schematics in chapter 7 "Application information" of the data sheet (basically 4 caps and some gound/route information)

From the homepage it sais: "A pull-up resistor may be required on the Chip/Slave Select line of the SCL3300. A typical resistor value of 4.7k ohms should connect this pin to +3.3 volts." and "Be sure to connect the SCL3300 DVIO pin to +3.3 volts as well. This pin powers the digital pins."

So... from that I gather:
VDD -> 3.3v ESP32
GND -> GND ESP32
DVIO -> 3.3v ESP32
SCK, MOSI, MISO -> SCK, MOSI, MISO of ESP32
CSB -> 4.7k ohm -> CS GPIO or 3.3v.  -- shouldn't this just be a pullup resistor?

That seems correct to you?

However, the manual also sais for the SPI inputs: "3.3V logic compatible Schmitt-trigger input". ESP32 does not have that by default; and while all SPI should have that, it does leaves me wondering if I should add an 74AHCT14 (two times per line) for the SPI lines?

Any input would be appreciated!

DavidArmstrong, 2021-07-29:
Hi,
I thought I had seen an email come in, but it disappeared from my inbox before I could reply.  Sorry for the later-than-normal response.  Anyway, the chip select line has to be controlled by the MCU (ESP32 in this case.)  It should not just be tied high, as its transitions are used as part of the timing for SPI data transfers.  The pullup resistor on this line is needed in some cases to help that line transition high.  Not all systems will need it though.  I have one test system that requires the resistor, and the other system doesn't.
As far as using Schmitt trigger inputs, I have found that they can help in certain cases, and so I'm not adverse to including them.  I know that the manufactured PCB board is a bit on the expensive side, and I've heard various reasons for that.  (It's not to try to get more money out of it, believe it or not.)  However this particular chip has some special requirements for surface mount soldering, due to the manufacturing of the sensor itself.  So designing a one-off PCB is going an adventurous route, to be sure.

atlaste, 2021-07-30:
@DavidArmstrong Thanks for the fast and clear response.

We're going to see if it's hard to fab a board of my own, because I just got a few sensors. It's not that I don't want to pay a few bucks more for a perfectly pre-mounted pcb in this particular case - it's just that they're really unavailable in this side of the world. Shipping them from the other side of the world here costs more than the PCB itself, and a lot of time - while the IC's themselves actually are available. Also interesting is that the prototype PCB does not seem to conform to the mounting instructions - specifically the part about not putting traces at the back of the IC.. :-)

I've read the mounting instructions.. Yeap, they're going to be interesting. :) I'm just going to build the small PCB board that's in the datasheet with some pin headers - that way I can mess up without getting hurt too much. 

I'll let you know if it worked with some pics (for giggles).

atlaste, 2021-07-31:
First off, the library. It seems to work just fine. For ESP32 it is normal practice though to specify the pins. CS can be confusing here, since it's configured at two places. The proper way to do it is:

```c++
    SPI.begin(33, 27, 25, scl3300_csPin);
  // Maximum SPI frequency is 2 MHz - 4 MHz to achieve the best performance
  // initialize the chip select pin:
  pinMode(scl3300_csPin, OUTPUT);
  digitalWrite(scl3300_csPin, HIGH);
```

That being said, I'm very happy at the moment. So here's for giggles :-) 

[image]
[image]
[image]
[image]

I basically stole the design from the datasheet (figure 17), then milled a PCB according to that plan on my CNC. Did some tests first that I've been wanting to do for some time now (hence the blurry traces), and after that went for the real deal. Solder masked them. Then decided to go for it.

Well, not quite... The PCB's in the third screenshot are wrong. I was confused by the IC's; they have 12 holes on the bottom, and one of them is in the corner. Normally, these things are used to indicate pin 1. Well, this IC is apparently different... if you look at how they are railed and the assembly mounting guide, pin 1 is the opposite one. Fun times... So I started over, made some more PCB's.

I can *really* not recommend anyone to do this. I only used a heat gun to do the soldering. Used a CNC laser to make a solder mask in paper, then put liquid solder on top of it and then used an aluminium piece to put the solder on. That's when the real fun started... Solder mask usually extends and then contracts - so with IC's like this, you just have to pray it works. Moreover, keep them at the right temperature, and ramp at 3C/sec up and 6C/sec down. So, no heat gunning it and then moving away - I was careful to time it. Next, it did not work - so I had to rework it. Again heat it up, once at temperature use a plier to lift it, check what's wrong and carefully put it down again. 

I *think* it can be done easier by first soldering the contacts and then doing a separate ramp up/down cycle while applying a bit of pressure on the IC. Since I have a second IC that I have to solder in the next 72-ish hours, I guess we'll find out :-)

Test. Works!

As I said, I'd rather just buy a dev board given the choice.

DavidArmstrong, 2021-07-31:
"Well, this IC is apparently different..."
LOL.  That's an understatement.  I suspect all the strangeness is due to the physics of making a low noise, accurate tilt sensor.  The other tilt sensors out there are nowhere near as sensitive, nor low noise.  That being said, I'll endure the quirks of this chip just to get that level of accuracy.  And as I wrote before, I knew your PCB path would end up being 'adventurous'.  I'm glad you have finally been able to have some success with this though.
Oh, and I did find a copy of the older Revision 1 of the datasheet that I used to write the library originally.  They obviously changed the recommended startup sequence with the Revision 2 document.  I incorporated those updates into the latest 3.1.0 release of the library.

atlaste, 2021-08-03:
"And as I wrote before, I knew your PCB path would end up being 'adventurous'. I'm glad you have finally been able to have some success with this though." - I would just call it a win. Both my PCB's ended up working perfectly. Perhaps one of them is tilted a bit or something like that - personally I don't care too much about that, since I'm mainly looking for relative tilt.

I did notice a bit of noise, but overall I am very, very impressed with these IC's.

I'm wondering, since you have a lot more experience with these than I have, I noticed that the accuracy is worse when you get near 90 degrees, which makes sense to me. I want to use these (among other things) to mount something as perfect as possible to 90 degrees. Would you create a fixture that has the sensor on 45 degrees for that? Or would you use the z axis for that?

DavidArmstrong, 2021-08-04:
If you want to measure specifically for something that is vertical, I would make something so as to have the sensor be level at 90 degrees.  But otherwise, using the Z-axis should be fine.  It's the main reason why the sensor is as thick as it is.  However, it's a tricky thing to get something either truly flat, or truly vertical.  For example, my purpose for getting these sensor is for telescope alignment.  To see if the mount is truly level though, I rotate it through its 360 degree range going around, and see how the sensor measures level over the course of the circle.  That helps account for sensor mounting errors, which I always seem to have.

atlaste, 2021-08-05:
@DavidArmstrong Tell me about it... Measuring flatness is really something else...

I'm working on CNC machines as a little out-of-hand hobby, and want to use these sensors for variety of reasons. Some of these might work, some might not... 

- First off, my cnc is built around a DIY surface plate - and I want to measure how flat that is. I made a PCB with 3 little ball bearings below it, and want to measure inclination at a lot of points. I'm first attempting this manually to see if it works - if so, I'll hook it up to a midTbot to do more accurate measurements.
- Second, I want to measure the linear rails and bearings. This is quite tricky, and obviously I'm not only using this but also a measuring clock and some other instruments. Again, I mainly focus on flatness here.
- Third, I want to measure the squareness of the machine; for example, the spindle should mounted 90' measured from the surface plate and the side rails. A manual square just isn't good enough; so I use set screws. Basically I want to measure it, and finetune it until it's perfect.
- And last, I want to measure motion, specifically: backlash of the bearings, through the accelerometer. Basically move a mm, meaure, move a mm, etc. I think this is going to be quite challenging... 

If you have any suggestions or tips, they are highly appreciated.

atlaste, 2021-08-09:
@DavidArmstrong FYI, "The other tilt sensors out there are nowhere near as sensitive, nor low noise. "

https://www.analog.com/media/en/technical-documentation/data-sheets/adxl354_adxl355.pdf seems to indicate an even higher sensitivity.

DavidArmstrong, 2021-08-10:
I probably should have clarified my statement with something like "of all the sensors that I could obtain at that time." I'd expect there to be competing sensors available, but back when I started this project, it was the best of what I could find. (Look at how many different temperature sensors are available now. ) 
Glad you shared that link with me though. Thanks!

DavidArmstrong, 2022-03-06:
Closing issue after being open for more than six months with no further comments or updates.

closed 2022-03-06 by DavidArmstrong