DavidArmstrong/SCL3300 #31: Accuracy in Mode 4

tamasharasztosi, 2022-01-11:
Hello!

I would like to use this sensor in Mode 4, 10 Hz mode (I know it is the default mode) however when I use your code without delay(100) I get around 250 Hz data.
I get good values however it jumps around +/-3 in RAW values which is too much for my application.

How can I correctly initiate 10 Hz low noise mode?

Thank you!

DavidArmstrong, 2022-01-11:
The data sheet states this caveat regarding the orientation of the sensor in modes 3 and 4:

"The orientation in which the Y-axis is parallel to gravity (i.e. ±1g) is not recommended when using either mode 3 or mode 4."

So, for mode 4, the Y-axis has to be pointing *horizontally*.  Since this is the orientation you state you are using, this should be fine.  Also, the '10 Hz' number listed in the datasheet is for stating the response time of the first pass filter internal to the sensor in mode 4 - it is not the speed of reading the data in mode 4.  Those are two different things.

Now then, to reduce the effect of noise, the easiest and best way to do this is to take a number of readings and average them.  Using the sensor fast read mode, as you have stated, gives a very high number of readings per second.  If a number of these values are read as a group, and averaged together as one value, then that will reduce the noise.  Of course, it will also reduce the number of 'data points' taken per second, but that is the trade-off between accuracy and speed.

You will need to define a 'for' loop to read in a number of values, say 10 or 25, and average them.  (i.e. One average number for the X-axis, a separate average for the Y-axis, and a third average for the Z-axis.)  All three averages can be computed on the same set of data being read in, so you don't need three different 'for' loops to do it.

tamasharasztosi, 2022-01-11:
Okay!

Yes I will do the good old oversampling. 
Thank you for your reply!

DavidArmstrong, 2023-01-29:
No further comments in over a year. Closing issue.

closed 2023-01-29 by DavidArmstrong