DavidArmstrong/SCL3300 #32: reset() function

tamasharasztosi, 2022-02-03:
Hello!

There is SCL3300::reset(void) function that has this line:

> begin(); //Re-init chip

If chipselect pin is not the default then it's going to be failed as I experience.

DavidArmstrong, 2022-02-03:
Hello,

If the chip select pin has been changed from the default value by an Initial call to begin(csPin), then subsequent calls internally will not change this, nor reference the default value. I'll run a test this weekend to  verify and provide evidence of this.
Dave

DavidArmstrong, 2022-02-06:
The sample sketch Example8_TwoSensors shows the library functioning with two SCL3300 sensors at the same time.  (One is using the default Chip Select line on D10, and the other on D9).  Since I actually have two sensors, I have been able to test this sketch, and it works fine.  Likewise, if I change Example1_BasicTiltLevelOffset so it just talks to the SCL3300 sensor on the D9 line, it still works just fine.  (There are positions where moving my sensor makes it 'freeze' momentarily, but I suspect that is more due to mounting it on a breadboard rather than having a permanent soldered setup. SPI is highly sensitive to any communication glitches.)

tamasharasztosi, 2022-02-06:
Thank you for testing this out!
As further I tested the initial set of cspin is works fine however SPI is the reason of my problems. 
It just cant open SPI bus at times, however if I removed else reset() part the code doesnt stop, I guess the reinit (with begin()) of the sensor takes more time.

Do you have any suggestion for SPI sensitivity?

DavidArmstrong, 2022-02-07:
While removing reset() will keep the code from freezing, what is happening under the hood is that it enters an error state, and stays there until a reset() is done.

One hint I can give is as follows:  The data sheet states this caveat regarding the orientation of the sensor in modes 3 and 4:

"The orientation in which the Y-axis is parallel to gravity (i.e. ±1g) is not recommended when using either mode 3 or mode 4."

So, for mode 4, the Y-axis has to be pointing horizontally for best performance.  I suspect that this is also affecting when the sensor needs to be reset.

DavidArmstrong, 2023-01-29:
No update in nearly a year. Closing issue.

closed 2023-01-29 by DavidArmstrong