DavidArmstrong/SCL3300 #36: 3 sensors, 3 different behaviours

fr4cc4, 2024-05-28:
Hi Dave, 
first of all thank you very much for this library, it was very helpful and it is very well written.
I am having some doubts though about how the sensors work. 
I purchased 3 sensors with pcb and connected them like this to my elegoo nano:
CS - pin D10
MOSI - pin D11
MISO - pin D12
SCK - pin D13
AVSS - GND
AVDD - 3v3
One sensor seems to be working properly (although it disconnects in severe bumps), the second sensor seems to be burned out (by activating debugging it seems not to be connected) the third stopped working properly after a few hours of use.
The third sensor returns almost fixed and wrong values for both accelerations, angles and temperature.
I also tried connecting the DVIO and adding a pull-up resistor on CS as you suggest but the situation does not change.
Do you happen to have any suggestions ? Is there any command in the library that can help me ?
Thank you very much,
Francesco

DavidArmstrong, 2024-06-07:
First, I apologize for taking so long to get back to you.  Second, if three sensors are being used on the same system, then each sensor must have a different Arduino pin connected to their CS line.  So, having one SCL3300 CS pin connected to pin 10 on the Arduino, the next CS going to pin 9, and the third CS line going to pin 8 would work.  ( The final choices are yours, of course. )
Then one just has to make the correct call to select the device being accessed.

fr4cc4, 2024-06-08:
Hi Dave,
no problem, thanks for your answer.
I misspoke, right now I'm using the same Arduino Nano to test the 3 sensors but I'm connecting them one at a time.
My doubt is on the hardware configuration: I'm not sure if I **must** connect DVIO pin to 3.3V and/or if I **must** use the pull-up resistor to make the sensors work properly.
I received other 2 new sensors and the behave as the "working" one: in case of bumps, even very light, the disconnect from arduino, like for 1 or 2 iteration. Do you ever experienced anything like that ? Could this behaviour be related to DVIO missing connection ?
Thanks in advance.
Looking forward to your response,
Francesco

DavidArmstrong, 2024-06-09:
Francesco,

Another point to consider is the mode the sensor is put into. Mode 4 is
only rated for use +-10 degrees from level, and I've found issues outside
of that range. I've had to switch to mode 0 or 1 in that case, which works
much better when the tilt can get large.

Oh, and DVIO must be connected to a 3.3 volt power source regardless of the
application.

David

fr4cc4, 2024-06-10:
Thanks again for your help, I'll try using mode 0 and 1 and also connecting DVIO. Is it ok to connect to the same 3v3 PIN of AVDD ?

fr4cc4, 2024-06-24:
If I connect also DVIO pin and I check voltage between Arduino Nano pin 3v3 and GND I get between 6V and 7V. Is it right ? maybe I misunderstood something. 
I cannot understand why I get reasonable measurements with both the pin connected and the pin disconnected. Is it mandatory to connect DVIO pin ?

Arakistech, 2024-07-12:
Do you think this library would be usable with Teensy 4.1?

Regards,

DavidArmstrong, 2024-07-13:
For the DVIO pin, this is what the datasheet shows:
[image]

Also, I've used the Sparkfun Miromod platform for testing, with their Teensy processor.  So using Teensy 4.1 will work fine.

Dave

DavidArmstrong, 2025-12-09:
No further responses in over a year.  Closing issue.